fdu/STM32F746G-disco_Buildroot #1: ethernet

trnila, 2018-03-18:
Hello,
I am trying to get ethernet working, but I have some issue with the following device tree configuration.
```
#include <dt-bindings/pinctrl/stm32f746-pinfunc.h>
#include "stm32f746-disco.dts"
/ {
	reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;

		linux,dma {
			compatible = "shared-dma-pool";
			linux,dma-default;
			no-map;
			reg = <0x20010000 0x3bfff>; /* sram1 240kb */
		};
	};

	soc {
		mac: ethernet@40028000 {
			compatible = "st,stm32-dwmac", "snps,dwmac-3.50a", "snps,dwmac";
			reg = <0x40028000 0x8000>;
			reg-names = "stmmaceth";

			resets = <&rcc STM32F7_AHB1_RESET(ETHMAC)>;
			reset-names = "stmmaceth";

			clocks = <&rcc 0 STM32F7_AHB1_CLOCK(ETHMAC)>,
				 <&rcc 0 STM32F7_AHB1_CLOCK(ETHMACTX)>,
				 <&rcc 0 STM32F7_AHB1_CLOCK(ETHMACRX)>;
			clock-names = "stmmaceth", "mac-clk-tx", "mac-clk-rx";

			interrupts = <61>, <62>;
			interrupt-names = "macirq", "eth_wake_irq";

			dma-ranges;
			st,syscon = <&syscfg 0x4>;

			pinctrl-0 = <&ethernet_rmii>;
			phy-mode = "rmii";
			phy-handle = <&phy0>;

			status = "okay";

			mdio0 {
				#address-cells = <1>;
				#size-cells = <0>;
				compatible = "snps,dwmac-mdio";
				phy0: ethernet-phy@0 {
					reg = <0>;
				};
			};
		};
	};
};

&pinctrl {
	ethernet_rmii: rmii@0 {
	      pins {
		      pinmux = <STM32F746_PG13_FUNC_ETH_MII_TXD0_ETH_RMII_TXD0>,
			     <STM32F746_PG14_FUNC_ETH_MII_TXD1_ETH_RMII_TXD1>,
			     <STM32F746_PG11_FUNC_ETH_MII_TX_EN_ETH_RMII_TX_EN>,
			     <STM32F746_PA2_FUNC_ETH_MDIO>,
			     <STM32F746_PC1_FUNC_ETH_MDC>,
			     <STM32F746_PA1_FUNC_ETH_MII_RX_CLK_ETH_RMII_REF_CLK>,
			     <STM32F746_PA7_FUNC_ETH_MII_RX_DV_ETH_RMII_CRS_DV>,
			     <STM32F746_PC4_FUNC_ETH_MII_RXD0_ETH_RMII_RXD0>,
			     <STM32F746_PC5_FUNC_ETH_MII_RXD1_ETH_RMII_RXD1>;
		      slew-rate = <2>;
[... 106 lines cut ...]
[    1.300000] i2c /dev entries driver
[    1.310000] NET: Registered protocol family 10
[    1.330000] Segment Routing with IPv6
[    1.330000] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    1.340000] hctosys: unable to open rtc device (rtc0)
[    1.360000] Freeing unused kernel memory: 276K
[    1.360000] This architecture does not have kernel memory protection.
Initializing random number generator... urandom start: failed.
```

When I try to bring ethernet up, it fails on resetting ethernet's dma (https://github.com/torvalds/linux/blob/b9849860675f925da0380f4ea76c3f5041909737/drivers/net/ethernet/stmicro/stmmac/dwmac_lib.c#L35).
Bit 0 SR of register ETH_DMABMR is never cleared, so reset haven't completed (even when I increased timeout slightly).

```
~ # ifconfig eth0 up
[   83.410000] stm32-dwmac 40028000.ethernet eth0: device MAC address 46:12:3a:b2:7f:dd
[   83.560000] Generic PHY stmmac-0:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=stmmac-0:00, irq=POLL)
[   84.580000] stm32-dwmac 40028000.ethernet: Failed to reset the dma
[   84.580000] stm32-dwmac 40028000.ethernet eth0: stmmac_hw_setup: DMA engine initialization failed
[   84.590000] stm32-dwmac 40028000.ethernet eth0: stmmac_open: Hw setup failed
ifconfig: SIOCSIFFLAGS: Device or resource busy
```

Do you see any problem?
Thank you

cfriedt, 2018-12-14:
Ok - I am duplicating this problem now. But my stm32f746-disco.dtsi is

{noformat}
#include "skeleton.dtsi"
#include "armv7-m.dtsi"
#include <dt-bindings/pinctrl/stm32-pinfunc.h>
#include <dt-bindings/clock/stm32fx-clock.h>
#include <dt-bindings/mfd/stm32f7-rcc.h>

/ {
	reserved-memory {
		#address-cells = <1>;
		#size-cells = <1>;
		ranges;

		linux,dma {
			compatible = "shared-dma-pool";
			linux,dma-default;
			no-map;
			reg = <0x20010000 0x3bfff>; /* sram1 240kb */
		};
	};
	clocks {
		clk_hse: clk-hse {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <0>;
		};

		clk-lse {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <32768>;
		};

		clk-lsi {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <32000>;
		};

		clk_i2s_ckin: clk-i2s-ckin {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <48000000>;
		};
	};

	soc {
		mac: ethernet@40028000 {
			compatible = "st,stm32-dwmac", "snps,dwmac-3.50a", "snps,dwmac";
			reg = <0x40028000 0x8000>;
			reg-names = "stmmaceth";

			resets = <&rcc STM32F7_AHB1_RESET(ETHMAC)>;
			reset-names = "stmmaceth";

			clocks = <&rcc 0 STM32F7_AHB1_CLOCK(ETHMAC)>,
				 <&rcc 0 STM32F7_AHB1_CLOCK(ETHMACTX)>,
				 <&rcc 0 STM32F7_AHB1_CLOCK(ETHMACRX)>;
			clock-names = "stmmaceth", "mac-clk-tx", "mac-clk-rx";

			interrupts = <61>, <62>;
			interrupt-names = "macirq", "eth_wake_irq";
			dma-ranges;
			st,syscon = <&syscfg 0x4>;

			pinctrl-0 = <&ethernet_rmii>;
			phy-mode = "rmii";
			phy-handle = <&phy0>;
[... 732 lines cut ...]
			compatible = "st,stm32f7-hsotg";
			reg = <0x40040000 0x40000>;
			interrupts = <77>;
			clocks = <&rcc 0 STM32F7_AHB1_CLOCK(OTGHS)>;
			clock-names = "otg";
			status = "disabled";
		};

		usbotg_fs: usb@50000000 {
			compatible = "st,stm32f4x9-fsotg";
			reg = <0x50000000 0x40000>;
			interrupts = <67>;
			clocks = <&rcc 0 STM32F7_AHB2_CLOCK(OTGFS)>;
			clock-names = "otg";
			status = "disabled";
		};
	};
};

&systick {
	clocks = <&rcc 1 0>;
	status = "okay";
};

{noformat}

cfriedt, 2018-12-23:
making some progress here

https://github.com/torvalds/linux/compare/master...cfriedt:cfriedt/stm32/get-ethernet-working?expand=1

```
~ # ifconfig eth0 up 192.168.1.250
[   11.365572] stm32-dwmac 40028000.ethernet eth0: device MAC address 92:27:a9:78:7e:e1
[   11.376138] Generic PHY stmmac-0:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=stmmac-0:00, irq=POLL)
[   11.407535] stm32-dwmac 40028000.ethernet eth0: No Safety Features support found
[   11.415266] stm32-dwmac 40028000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   11.426347] stm32-dwmac 40028000.ethernet eth0: registered PTP clock
[   11.434405] stmmac: (ptrval) <= 0041848c ETH_MACCR
[   11.439460] stmmac: (ptrval) <= 00000010 ETH_MACMIIAR
[   11.444527] stmmac: (ptrval) <= 00003100 ETH_MACMIIDR
[   11.449848] stmmac: (ptrval) <= 00000200 ETH_MACIMR
[   11.454745] stmmac: (ptrval) <= 8000e17e ETH_MACA0HR
[   11.459976] stmmac: (ptrval) <= 78a92792 ETH_MACA0LR
[   11.464950] stmmac: (ptrval) <= 00000024 ETH_MMCCR
[   11.470008] stmmac: (ptrval) <= 00ffffff ETH_MMCRIMR
[   11.474987] stmmac: (ptrval) <= 01ffffff ETH_MMCTIMR
[   11.480229] stmmac: (ptrval) <= 20010000 ETH_DMARDLAR
[   11.485294] stmmac: (ptrval) <= 20014000 ETH_DMATDLAR
[   11.490616] stmmac: (ptrval) <= 02202006 ETH_DMAOMR
[   11.495507] stmmac: (ptrval) <= 0001a061 ETH_DMAIER
[   11.503339] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
~ # [   12.491707] stm32-dwmac 40028000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   12.500698] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   18.409153] ------------[ cut here ]------------
[   18.413906] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:461 dev_watchdog+0xfd/0x19c
[   18.422102] NETDEV WATCHDOG: eth0 (stm32-dwmac): transmit queue 0 timed out
[   18.429135] CPU: 0 PID: 0 Comm: swapper Not tainted 4.19.0 #33
[   18.434961] Hardware name: STM32 (Device Tree Support)
[   18.440295] [<c000c0c9>] (unwind_backtrace) from [<c000b223>] (show_stack+0xb/0xc)
[   18.448044] [<c000b223>] (show_stack) from [<c000eb0f>] (__warn+0x87/0xa0)
[   18.455093] [<c000eb0f>] (__warn) from [<c000eb4f>] (warn_slowpath_fmt+0x27/0x40)
[   18.462773] [<c000eb4f>] (warn_slowpath_fmt) from [<c01a450d>] (dev_watchdog+0xfd/0x19c)
[   18.471064] [<c01a450d>] (dev_watchdog) from [<c0037b49>] (call_timer_fn+0xf/0x56)
[   18.478821] [<c0037b49>] (call_timer_fn) from [<c0037ccb>] (expire_timers+0x7b/0x7e)
[   18.486758] [<c0037ccb>] (expire_timers) from [<c0037ecd>] (run_timer_softirq+0xf5/0x140)
[   18.495125] [<c0037ecd>] (run_timer_softirq) from [<c000994d>] (__do_softirq+0xf5/0x14c)
[   18.503402] [<c000994d>] (__do_softirq) from [<c001098d>] (irq_exit+0x3d/0x80)
[   18.510812] [<c001098d>] (irq_exit) from [<c002be75>] (__handle_domain_irq+0x47/0x6a)
[   18.518830] [<c002be75>] (__handle_domain_irq) from [<c000ba83>] (__irq_entry+0x53/0x84)
[   18.526947] ---[ end trace 0eb52e36aeb5afb5 ]---
```

cfriedt, 2019-01-13:
This patch works for me
[stm32f746-disco-get-ethernet-working.diff.gz](https://github.com/fdu/STM32F746G-disco_Buildroot/files/2753379/stm32f746-disco-get-ethernet-working.diff.gz)

cfriedt, 2019-01-13:
```
$ ping -c 100 192.168.1.250
PING 192.168.1.250 (192.168.1.250) 56(84) bytes of data.
64 bytes from 192.168.1.250: icmp_seq=1 ttl=64 time=6.61 ms
...
64 bytes from 192.168.1.250: icmp_seq=100 ttl=64 time=4.92 ms

--- 192.168.1.250 ping statistics ---
100 packets transmitted, 100 received, 0% packet loss, time 99162ms
rtt min/avg/max/mdev = 1.710/4.397/10.945/1.265 ms
```

hussainali0935, 2019-02-26:
Greetings and Regards
I tried to apply this patch  and set up Ethernet. But unfortunately I could not get the right result.
The operating system will boot and I can assign IP to Ethernet interface, but stm32 will not answer ping. Also, stm32 can not ping host computer.
Also, because you did not mention this way of configuring kernel to set up Ethernet in this thread, so I made the settings myself in the kernel, which, of course, do not know if it is enough or not.
Thank you for helping me.

This is my kernel config and dts files:
[configs.zip](https://github.com/fdu/STM32F746G-disco_Buildroot/files/2905849/configs.zip)

cfriedt, 2019-03-02:
> Also, because you did not mention this way of configuring kernel to set up Ethernet in this thread, so I made the settings myself in the kernel, which, of course, do not know if it is enough or not.

Strange.

After applying these patches, I rebuilt my kernel with
`make ARCH=arm CROSS_COMPILE=arm-none-eabi-  stm32_defconfig`
`make ARCH=arm CROSS_COMPILE=arm-none-eabi-`

and have not had any issues (although dma still does not work with uart / i2c)

You have console access on ttySTM0, right?

hussainali0935, 2019-03-06:
Thanks for quick answering.

My console is on ttyACM0.

I used the master branch of repo  where the kernel is configure with a file located in the configs dir( named 'linux') . And since I did not have Ethernet settings, I had to change and add the Ethernet settings to kernel.

What version of the kernel are you using?

After reading your answer, I changed the kernel configuration file and used the 'stm32_defconfig' file.
The result was that it did not compile  (due to 'XIP' option that was enabled in kernel). After disabling this option, kernel compiled successfully but evaluation board does not boot  and kernel stuck at 
 message 'starting kernel ... '.

cfriedt, 2019-03-07:
Sorry - my bad - I should have clarified that the STM32 will boot linux and use ttySTM0 as the console.

My exact config is here.

[linux-stm32f746.config.txt](https://github.com/fdu/STM32F746G-disco_Buildroot/files/2942595/linux-stm32f746.config.txt)

Other than that, everything else should be in the branch

https://github.com/cfriedt/linux/compare/branch-v4.20...cfriedt:cfriedt/stm32f746-disco/get-ethernet-working

Note, the part that says 

```
 I've noticed that U-Boot bootargs trump these
		 * I suggest adding the following to your U-Boot .config
		 * CONFIG_BOOTARGS="console=ttySTM0,115200n8"
```

sadegh916, 2019-11-14:
Can somebody help me step by step to enable ethernet for this board?

rodrigo455, 2021-02-09:
I was able to overcome this problem with the following patch:

```diff
 drivers/net/ethernet/stmicro/stmmac/dwmac-stm32.c | 8 +++++---
 1 file changed, 5 insertions(+), 3 deletions(-)

diff --git a/drivers/net/ethernet/stmicro/stmmac/dwmac-stm32.c b/drivers/net/ethernet/stmicro/stmmac/dwmac-stm32.c
index 9e6db16af..63d6f388b 100644
--- a/drivers/net/ethernet/stmicro/stmmac/dwmac-stm32.c
+++ b/drivers/net/ethernet/stmicro/stmmac/dwmac-stm32.c
@@ -22,7 +22,9 @@
 
 #include "stmmac_platform.h"
 
-#define MII_PHY_SEL_MASK	BIT(23)
+#define STM32_SYSCFG_PMC_ETHMODE_MASK		(0x3 << 23)
+#define STM32_SYSCFG_PMC_ETHMODE_MII		(0x0 << 23)
+#define STM32_SYSCFG_PMC_ETHMODE_RMII		(0x1 << 23)
 
 struct stm32_dwmac {
 	struct clk *clk_tx;
@@ -39,8 +41,8 @@ static int stm32_dwmac_init(struct plat_stmmacenet_data *plat_dat)
 	u32 val;
 	int ret;
 
-	val = (plat_dat->interface == PHY_INTERFACE_MODE_MII) ? 0 : 1;
-	ret = regmap_update_bits(dwmac->regmap, reg, MII_PHY_SEL_MASK, val);
+	val = (plat_dat->interface == PHY_INTERFACE_MODE_MII) ? STM32_SYSCFG_PMC_ETHMODE_MII : STM32_SYSCFG_PMC_ETHMODE_RMII;
+	ret = regmap_update_bits(dwmac->regmap, reg, STM32_SYSCFG_PMC_ETHMODE_MASK, val);
 	if (ret)
 		return ret;
 
-- 
```

mbasilSB, 2023-01-30:
Hi, I applied both patches, @cfriedt and @rodrigo455, with the configuration of @cfriedt , and got stuck in the following part in both of them:
_~ # ifconfig eth0 up 192.168.0.123
[   50.320000] stm32-dwmac 40028000.ethernet eth0: device MAC address b2:09:eb:cc:42:88
[   50.530000] Generic PHY stmmac-0:00: attached PHY driver [Generic PHY] (mii_bus:phy_addr=stmmac-0:00, irq=POLL)
[   50.550000] stm32-dwmac 40028000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   50.560000] stm32-dwmac 40028000.ethernet eth0: registered PTP clock
[   50.560000] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
~ # [   52.010000] stm32-dwmac 40028000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[   52.010000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   57.440000] ------------[ cut here ]------------
[   57.440000] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:323 dev_watchdog+0xfd/0x19c
[   57.440000] NETDEV WATCHDOG: eth0 (stm32-dwmac): transmit queue 0 timed out
[   57.440000] CPU: 0 PID: 0 Comm: swapper Not tainted 4.15.7 #2
[   57.440000] Hardware name: STM32 (Device Tree Support)
[   57.440000] [<c000bfc5>] (unwind_backtrace) from [<c000b34f>] (show_stack+0xb/0xc)
[   57.440000] [<c000b34f>] (show_stack) from [<c000e6ab>] (__warn+0x87/0xa0)
[   57.440000] [<c000e6ab>] (__warn) from [<c000e6e3>] (warn_slowpath_fmt+0x1f/0x28)
[   57.440000] [<c000e6e3>] (warn_slowpath_fmt) from [<c0183eb9>] (dev_watchdog+0xfd/0x19c)
[   57.440000] [<c0183eb9>] (dev_watchdog) from [<c0034bc1>] (call_timer_fn+0xf/0x56)
[   57.440000] [<c0034bc1>] (call_timer_fn) from [<c0034c83>] (expire_timers+0x7b/0x7e)
[   57.440000] [<c0034c83>] (expire_timers) from [<c0034d65>] (run_timer_softirq+0xdf/0x11e)
[   57.440000] [<c0034d65>] (run_timer_softirq) from [<c0009323>] (__do_softirq+0xf3/0x14c)
[   57.440000] [<c0009323>] (__do_softirq) from [<c001043f>] (irq_exit+0x6b/0x7c)
[   57.440000] [<c001043f>] (irq_exit) from [<c002a205>] (__handle_domain_irq+0x47/0x6a)
[   57.440000] [<c002a205>] (__handle_domain_irq) from [<c000b9a7>] (__irq_entry+0x53/0x84)
[   57.440000] [<c000b9a7>] (__irq_entry) from [<00000006>] (0x6)
[   57.440000] ---[ end trace 110a9d61914add72 ]---**_

if after this if I try ifconfig or ping I get the following error output:

~ # ifconfig
[  199.390000] ifconfig: page allocation failure: order:7, mode:0x14000c0(GFP_KERNEL), nodemask=(null)
[  199.390000] CPU: 0 PID: 57 Comm: ifconfig Tainted: G        W        4.15.7 #2
[  199.390000] Hardware name: STM32 (Device Tree Support)
[  199.390000] [<c000bfc5>] (unwind_backtrace) from [<c000b34f>] (show_stack+0xb/0xc)
[  199.390000] [<c000b34f>] (show_stack) from [<c00462b7>] (warn_alloc+0x51/0xce)
[  199.390000] [<c00462b7>] (warn_alloc) from [<c0046711>] (__alloc_pages_nodemask+0x391/0x564)
[  199.390000] [<c0046711>] (__alloc_pages_nodemask) from [<c00468ef>] (__get_free_pages+0xb/0x20)
[  199.390000] [<c00468ef>] (__get_free_pages) from [<c0046b0d>] (alloc_pages_exact+0x19/0x22)
[  199.390000] [<c0046b0d>] (alloc_pages_exact) from [<c00525c7>] (do_mmap+0x41d/0x5be)
[  199.390000] [<c00525c7>] (do_mmap) from [<c004e6cf>] (vm_mmap_pgoff+0x4b/0x60)
[  199.390000] [<c004e6cf>] (vm_mmap_pgoff) from [<c007d9ad>] (load_flat_file+0x375/0x552)
[  199.390000] [<c007d9ad>] (load_flat_file) from [<c007dbd3>] (load_flat_binary+0x49/0x2a6)
[  199.390000] [<c007dbd3>] (load_flat_binary) from [<c005bb63>] (search_binary_handler+0x57/0xe4)
[  199.390000] [<c005bb63>] (search_binary_handler) from [<c005be43>] (do_execveat_common+0x253/0x368)
[  199.390000] [<c005be43>] (do_execveat_common) from [<c005bf6d>] (do_execve+0x15/0x1a)
[  199.390000] [<c005bf6d>] (do_execve) from [<c0009a81>] (ret_fast_syscall+0x1/0x58)
[  199.390000] warn_alloc_show_mem: 1 callbacks suppressed
[  199.390000] Mem-Info:
[  199.390000] active_anon:0 inactive_anon:0 isolated_anon:0
[  199.390000]  active_file:0 inactive_file:0 isolated_file:0
[  199.390000]  unevictable:172 dirty:0 writeback:0 unstable:0
[  199.390000]  slab_reclaimable:32 slab_unreclaimable:563
[  199.390000]  mapped:0 shmem:0 pagetables:0 bounce:0
[  199.390000]  free:249 free_pcp:0 free_cma:0
[  199.390000] Node 0 active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:688kB isolated(anon):0kB isolated(file):0kB mapped:0kB dirty:0kB writeback:0kB shmem:0kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[  199.390000] Normal free:996kB min:272kB low:340kB high:408kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:688kB writepending:0kB present:8192kB managed:4964kB mlocked:0kB kernel_stack:200kB pagetables:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[  199.390000] lowmem_reserve[]: 0 0
[  199.390000] Normal: 1*4kB (U) 0*8kB 2*16kB (U) 0*32kB 1*64kB (U) 1*128kB (U) 3*256kB (U) 0*512kB 0*1024kB 0*2048kB 0*4096kB = 996kB
[  199.390000] 173 total pagecache pages
[  199.390000] 2048 pages RAM
[  199.390000] 0 pages HighMem/MovableOnly
[  199.390000] 807 pages reserved
[  199.390000] nommu: Allocation of length 286720 from process 57 (ifconfig) failed
[  199.390000] active_anon:0 inactive_anon:0 isolated_anon:0
[  199.390000]  active_file:0 inactive_file:0 isolated_file:0
[  199.390000]  unevictable:172 dirty:0 writeback:0 unstable:0
[  199.390000]  slab_reclaimable:32 slab_unreclaimable:563
[  199.390000]  mapped:0 shmem:0 pagetables:0 bounce:0
[  199.390000]  free:249 free_pcp:0 free_cma:0
[... cut ...] :200kB pagetables:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[  199.390000] lowmem_reserve[]: 0 0
[  199.390000] Normal: 1*4kB (U) 0*8kB 2*16kB (U) 0*32kB 1*64kB (U) 1*128kB (U) 3*256kB (U) 0*512kB 0*1024kB 0*2048kB 0*4096kB = 996kB
[  199.390000] 173 total pagecache pages
[  199.390000] binfmt_flat: Unable to allocate RAM for process text/data, errno -12
SEGV

Does anyone have a hint on why am I getting this error applying the exact same configuration and patches?

Thanks in advance

cfriedt, 2023-01-30:
It looks like the defines added in the above patch ate likely bogus. They define those symbols to nothing..

mbasilSB, 2023-01-30:
Thanks for the fast reply @cfriedt , I'm not sure I understood what yo mean, since I took the exact same configuration and version as yours and used your patch, I don't know what I may have done wrong in the process, do you have any hint about it or any possible solution?
Thanks

cfriedt, 2023-01-30:
> Thanks for the fast reply @cfriedt , I'm not sure I understood what yo mean, since I took the exact same configuration and version as yours and used your patch, I don't know what I may have done wrong in the process, do you have any hint about it or any possible solution?
> Thanks
> 

There is a patch listed above that someone claimed resolved their DMA issue. It is not mine. That's what I was referring to.

In terms of what you've done wrong, I have no idea. It's possible that source code has evolved since then. My comments applied to things 4 years ago but might be out of date now.

You may need to debug the issue yourself.