fdu/STM32F746G-disco_Buildroot #2: cant find my-stm32f746-disco.dts

Pve88, 2018-08-31:
when I try to make build, I see error:
>>> linux custom Building
cp -f ../configs/my-stm32f746-disco.dts /home/pve/Progs/stm32br/STM32F746G-disco_Buildroot/buildroot/output/build/linux-custom/arch/arm/boot/dts/
cp: cannot stat '../configs/my-stm32f746-disco.dts': No such file or directory

So file my-stm32f746-disco.dts doesnt exists

popy2k14, 2020-02-06:
bump! Same issue here

popy2k14, 2020-02-08:
Got it to compile.

See forks there are some patches regarding building. Use these. 

Then Search stm32f746-disco.dts in the directory and copy the one from the kennel directory and copy it to. /configs/my-stm32f746-disco.dts

Don't miss the "my-" has been added at the beginning of the filename.

You also have to install atftp with apt-get install. 

The build with: sudo make build

Good luck

This works

TA1DB, 2020-02-08:
> Got it to compile.
>
> See forks there are some patches regarding building. Use these.
>
> Then Search stm32f746-disco.dts in the directory and copy the one from the kennel directory and copy it to. [...]

Hi @popy2k14 , thanks for the kind feedback and for the information provided.

As far as I could see @beattie 's fork was 2 commit ahead of fdu:master, I downloaded his fork and applied the patch of @nanobrain (removed -J from Makefile ) , installed atftp as you've recommended, @beattie 's fork was already including the /configs/my-stm32f746-disco.dts file so I didn't do anything else, make clean, make bootstrap, make build.. , these files were created 

/home/user/STM32F746G-disco_Buildroot-master/buildroot/output/images/my-stm32f746-disco.dtb
/home/user/STM32F746G-disco_Buildroot-master/buildroot/output/images/rootfs.cpio
/home/user/STM32F746G-disco_Buildroot-master/buildroot/output/images/u-boot.bin
/home/user/STM32F746G-disco_Buildroot-master/buildroot/output/images/u-boot-spl.bin
/home/user/STM32F746G-disco_Buildroot-master/buildroot/output/images/zImage

But the build process ended with these :

Kernel: arch/arm/boot/zImage is ready
# Copy the kernel image(s) to its(their) final destination
/usr/bin/install -m 0644 -D /home/murat/STM32F746G-disco_Buildroot-master/buildroot/output/build/linux-custom/arch/arm/boot/zImage /home/murat/STM32F746G-disco_Buildroot-master/buildroot/output/images/zImage
# If there is a .ub file copy it to the final destination
test ! -f /home/murat/STM32F746G-disco_Buildroot-master/buildroot/output/build/linux-custom/arch/arm/boot/zImage.ub || cp /home/murat/STM32F746G-disco_Buildroot-master/buildroot/output/build/linux-custom/arch/arm/boot/zImage.ub /home/murat/STM32F746G-disco_Buildroot-master/buildroot/output/images
make[1]: Leaving directory '/home/murat/STM32F746G-disco_Buildroot-master/buildroot'
cp buildroot/output/images/my-stm32f746-disco.dtb /srv/tftp/stm32f746/stm32f746-disco.dtb
cp: cannot create regular file '/srv/tftp/stm32f746/stm32f746-disco.dtb': No such file or directory
Makefile:22: recipe for target 'build' failed
make: *** [build] Error 1

Removing the -J patch didn't make any change. I'm working on..

popy2k14, 2020-02-08:
Nice that you made progress.
You have to create the directory with:

`sudo mkdir /srv/tftp/stm32f746/`

Rerun:

`sudo make build`

This should make it.

TA1DB, 2020-02-08:
> Nice that you made progress.
> You have to create the directory with:
> `sudo mkdir /srv/tftp/stm32f746/`
> Rerun:
> `sudo make build`
> This should make it.

It worked, thanks a lot.. Could you please advise how can I proceed from here; "$ make flash_bootloader" process also completed without any error but nothing changes on the screen! (I tried pressing blue button at power-on reset) 
It seems that I can connect with tftp as seen below but I think it is fake because any ip address also connects! (seen at the buttom)
~$ tftp 192.168.201.6
tftp> status
Connected to 192.168.201.6.
Mode: netascii Verbose: off Tracing: off
Rexmt-interval: 5 seconds, Max-timeout: 25 seconds
tftp> 

tftp> connect 192.168.200.13 
tftp> status
Connected to 192.168.200.13.
Mode: netascii Verbose: off Tracing: off
Rexmt-interval: 5 seconds, Max-timeout: 25 seconds